StefanSchippers/xschem_sky130 #1: Error running xshem from xschem_sky130 directory

abhisupa, 2021-03-08:
Hi Stefan,

I am seeing following error while running xschem from xschem_sky130 direction
my directory has following files
[image]

Output error is 
[image]

Thanks and regards,
Abhishek

StefanSchippers, 2021-03-08:
Hi, you installed xschem by doing sudo apt install, but this installs a **very** old ubuntu pre built package. 
You need to install xschem from sources, by doing ./configure, make, sudo make install.

abhisupa, 2021-03-08:
Hi Stefan, 
I am running into following issue.
[image]

StefanSchippers, 2021-03-08:
just do ./bin/xschem

closed 2021-03-09 by abhisupa