StefanSchippers/xschem_sky130 #3: patch problem for sky130 in xschem

ripudamank2, 2021-04-19:
Hi, I was trying to run the patch command but getting this below error .
I already have openalne install in my system , so i was using this patch command in my openlane directory where pdks are install.
Can you please let me know how to solve this issue ? 

atch -p2 < /home/embedded/.xschem/xschem_library/xschem_sky130/sky130_fd_pr.patch
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -Naur sky130_fd_pr/latest/cells/nfet_01v8/sky130_fd_pr__nfet_01v8__ff.pm3.spice sky130_fd_pr_ngspice/latest/cells/nfet_01v8/sky130_fd_pr__nfet_01v8__ff.pm3.spice
|--- sky130_fd_pr/latest/cells/nfet_01v8/sky130_fd_pr__nfet_01v8__ff.pm3.spice	2020-11-08 01:33:06.385642949 +0100
|+++ sky130_fd_pr_ngspice/latest/cells/nfet_01v8/sky130_fd_pr__nfet_01v8__ff.pm3.spice	2020-12-08 04:28:01.986577051 +0100
--------------------------
File to patch: 

[image]

StefanSchippers, 2021-04-20:
you are in the `latest/` directory. 
Is there a '`cells/`' directory therein?
is there a `nfet_01v8/sky130_fd_pr__nfet_01v8__ff.pm3.spice` under '`cells/`' ?

ripudamank2, 2021-04-20:
I understood you point . 
There was nothing in that directory . So i did fresh installation of sky-water . Now its working .

thank you for the help.

closed 2021-05-03 by StefanSchippers