StefanSchippers/xschem_sky130 #4: Directions to import other librariers (spice or verilog)

leochand101, 2021-07-09:
I see there are some import scripts (verilog,spice) in the folders, but a set of directions to import other libraries will be helpful. 
Kindly point to specific files in case I missed the same.

StefanSchippers, 2021-07-09:
The decred_hash_macro is a spice netlist -> xschem import example, requested by a user, 
Normally you don't create schematics from a synthetized design, as the schematic contains thousands of gates/flops and is unreadable.
However in some very specific cases you might want to do that.

Importing a spice netlist into xschem can not be a general 'push button' process.
1) spice netlists don't specify the direction of subcircuit ports, so some additional information needs to be inserted in the spice netlist
2) Spice netlists written from various tools have all kinds of weird naming conventions, in some cases it is very hard to figure if a net is a part of a bus or a single signal.
The Readme file explains the process, a sequence of automated import scripts and some manual work.

A totally different subject is drawing the set of standard cells , like the  **sky130_fd_sc_hd** high density library.
what has been done was to manualy draw the symbol for one representative of each logic function (and, nand, nor, flop etc), then use some script to derive the bigger sizes (nand_2, nand_4, nand_8...) from the base (small, nand_1) cell.
So, creation of a standard cell symbol library in xschem is not an  automated process. If someone wants to do that it would be great (there are very good python programmers here).

leochand101, 2021-07-09:
A group of us are working on some of these, I think we had some correspondence regarding spice import last year, So we would like to contribute a similar structure on ASAP7 (assuming there are no licensing issues). I would ask you more pointed questions as we go through the same. We want to create general rules based on names (A,B, C, IN etc for I/P, D,O,Y etc for O/P and so on for I/O) and also associate those with specific symbol GATE structures (based on functions/structures etc). Closing this for the time being.

closed 2021-07-09 by leochand101

leochand101, 2021-09-22:
> The decred_hash_macro is a spice netlist -> xschem import example, requested by a user,
> Normally you don't create schematics from a synthetized design, as the schematic contains thousands of [...]
**_>    The Readme file explains the process, a sequence of automated import scripts and some manual work._**
> A totally different subject is drawing the set of standard cells , like the **sky130_fd_sc_hd** high density library.
> what has been done was to manualy draw the symbol for one representative of each [...]

Stefan, 
Which readme file were you pointing here , Can you send me the path relative to the xschem_sky130
Regards,

StefanSchippers, 2021-09-22:
The Readme is [here](https://github.com/StefanSchippers/xschem_sky130/blob/main/decred_hash_macro/README.md): Try to go through it, if you have problems let me know. 
Importing a synthetized spice netlist is a task that requires always some manual intervention, there are so many changing details over time and design space...