StefanSchippers/xschem_sky130 #5: Resistor models contain an error. Fix provided.

armleo, 2021-09-11:
Schematic:
[image]

If I try to run it this happens.

NGSPICE Log:
```
ngspice -r "serdes/resisor_issue.raw" serdes/resisor_issue.spice
******
** ngspice-34 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2020, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Sat Mar  6 03:53:35 UTC 2021
******

Compatibility modes selected: hs a

Circuit: **.subckt resisor_issue

Error: unknown subckt: xr1 0 net1 sky130_fd_pr__res_generic_m1 w=1 l=1 mult=1 m=1
    Simulation interrupted due to error!

ngspice 1 -> 
```

Netlist (my install path is /opt/pdk_root, other models like transistors work):
```
**.subckt resisor_issue
V2 net1 GND vdd_volts
XR1 GND net1 sky130_fd_pr__res_generic_m1 W=1 L=1 mult=1 m=1
**** begin user architecture code

.temp 125
.param vdd_volts=1.65
.param vddio_volts=3.0

.control
tran 0.01ns 20ns
write
.endc

.param mc_mm_switch=0

.lib "/opt/pdk_root/sky130A/libs.tech/ngspice/sky130.lib.spice" ss

**** end user architecture code
**.ends
.GLOBAL GND
** flattened .save nodes
.end
```

1. The spiceprefix by default has to be R instead of X. After fixing, it complains that mult parameter does not exist in the model
2. I modified the symbol file to remove mult parameter and set the spiceprefix to R by default and now it works. See diff below, this change might need to be applied to other models.

```
diff res_generic_m1.sym /opt/pdk_root/open_pdks/sources/xschem_sky130/sky130_fd_pr/res_generic_m1.sym 
20c20
< format="@spiceprefix@name @pinlist sky130_fd_pr__@model W=@W L=@L m=@mult"
---
> format="@spiceprefix@name @pinlist sky130_fd_pr__@model W=@W L=@L mult=@mult m=@mult"
25c25
< spiceprefix=R
---
> spiceprefix=X
```

See: https://github.com/RTimothyEdwards/open_pdks/issues/159

StefanSchippers, 2021-09-11:
Thank you for the report. Corrected res_generic_m1 as you suggested.

closed 2021-09-11 by StefanSchippers

armleo, 2021-09-11:
The other models might require changes too. I just gave an example. If I get my time around I will update the other and will post it.

StefanSchippers, 2021-09-11:
Yes i see something is being changed on resistors and i need to update. SOme resistors had a 3rd bulk terminal, now it seems these 3rd terminals are gone...

armleo, 2021-09-11:
No, the 3rd bulk terminal is only for some models. The metal/li1 models don't have them

StefanSchippers, 2021-09-11:
i know, but now also the diffusion resistors res_generic_nd do no more have the 3rd terminal. At least this is the message i get from ngspice...

StefanSchippers, 2021-09-15:
All resistors fixed according to most recent open_pdks / skywater-pdk models.