StefanSchippers/xschem_sky130 #10: test_inv.sch references $::SKYWATER_MODEL but when you run a NGSPICE simulation it does not exist

armleo, 2022-04-01:
For some reason, the variable here errors out.

Screenshot:
[image]

Code:
https://github.com/StefanSchippers/xschem_sky130/blob/e34045bca1b14e405a366a6ab5464c48969c8971/sky130_tests/test_inv.sch#L117

StefanSchippers, 2022-04-02:
I assume you installed open_pdks, then, make sure you copy the xschemrc into the directory where you start xschem:
something like:
`cp .../share/pdk/sky130A/libs.tech/xschem/xschemrc .`
Before starting xschem set the PDK_ROOT variable:
export PDK_ROOT=.../share/pdk
In above examples replace the '...' with the actual path where your open_pdks is installed.

armleo, 2022-04-02:
Can this be documented somewhere? A lot of people ask the following questions: How to install XSCHEM, How to make a XSCHEM project directory (as they need to copy the .spiceinit and from your suggestion xschemrc).

Maybe it's documented somewhere, but I couldn't navigate there from README.md.

StefanSchippers, 2022-04-02:
The process is explained in this [video](https://www.youtube.com/watch?v=KgBLByOkJxA) (at 5:40) which is linked in the pinned posts in the #xschem slack channel of [skywater-pdk.](https://app.slack.com/client/T01699QAZBQ/C017P3RAD42/thread/C017P3RAD42-1648733462.348639)
There is one recent addition: if you installed open_pdks in an unusual place you have to setup a PDK_ROOT variable that tells xschem where open_pdks is installed.

armleo, 2022-04-15:
I just wish this was documented in the text form, so I could refer to it when people asked.

closed 2022-04-15 by armleo