StefanSchippers/xschem_sky130 #18: Underlying transistor-based schematics of sky130_stdcells

mhajimorad, 2023-11-18:
Hi @StefanSchippers ,

Regarding the xschem symbols provided for the `sky130_fd_sc_hd` standard cell library, is there a means by which someone can see a symbol's underlying transistor-based schematic?

Thank you!

StefanSchippers, 2023-11-18:
There is no (to my knowledge) schematic or picture showing the standard cell implementations. The standard cell instances point directly to a spice netlist.

I have reverse engineered some spice netlists of these standard cells to get the transistor level schematic, but this is a extremely boring work.

mhajimorad, 2023-11-18:
Hi @StefanSchippers ,

Thank you for your message.  

To provide some context regarding my original post:  I am teaching an introductory VLSI design course for undergraduate students at my university.  I was hoping to have my students see/learn what is "underneath" each standard cell symbol block (as opposed to simply taking them for granted).

Based on your message, is my understanding correct that by looking at the spice netlist to which a standard cell instance points to, someone can "see" all the constituent transistors and corresponding connections?  

And for reference purposes, are you able to provide some guidance on how someone can find the spice netlist of a standard cell instance given its name?  For instance, in your `test_stdcells.sch` there are instances of `dfrtp_1` and `nand2_1`.  Where can someone find their corresponding spice netlists (and corresponding file names)?

Thank you!

StefanSchippers, 2023-11-18:
Of course you can see the transistor netlist. Take for example the simplest one, `inv_2.sym`. You may look into the file:
`.../share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl` 
and locate the inv_2 subcircuit:

```
.SUBCKT sky130_fd_sc_hd__inv_2 A VGND VNB VPB VPWR Y
*.PININFO A:I VGND:I VNB:I VPB:I VPWR:I Y:O
MMIN1 Y A VGND VNB nfet_01v8 m=2 w=0.65 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
MMIP1 Y A VPWR VPB pfet_01v8_hvt m=2 w=1.0 l=0.15 mult=1 sa=0.265
+ sb=0.265 sd=0.28 topography=normal area=0.063 perim=1.14
.ENDS sky130_fd_sc_hd__inv_2
```

You see there is a cmos pair with gate terminals connected to A and drains connected to Y. Source and body of the N-channel connected to VGND/VNB (usually ground nodes) and P-channel source and body connected to VPWR/VPB (usually VCC)

mhajimorad, 2023-11-18:
Thank you @StefanSchippers !

To make sure that I understand the conventions used in the file:

Is there some sort of implied scaling factor for the transistor widths and lengths?  For instance, in your `inv_2.sym` example, are the widths and lengths assumed to be in units of micro-meters?

And when it comes to the meaning of m=2 in your `inv_2.sym` example, does it translate to equivalent NMOS and PMOS transistors whose widths are 2*0.65 micro-meters and 2*1.0 micro-meters, respectively?

Thank you!

StefanSchippers, 2023-11-18:
sky130 has a `scale=1e-6 `set in the model file. This means a transistor specified with W=1 and L=0.15 has width equal to 1um (1e-6m) and length equal to 150nm (0.15e-6m).

For the inv_2 subcircuit, m=2 means there are two devices working in parallel (m is a multiplier factor)
so the actual sizes you deduced are correct.

mhajimorad, 2023-11-18:
Thank you @StefanSchippers !

And is there a difference in meaning between the `m` parameter versus `mult` ?

Thank you!

StefanSchippers, 2023-11-18:
no, m and mult are the same thing. xschem sets them to the same number. `mult` is simply needed because `mult` (like any generic parameter)  is available from inside the spice models, while `m` is not. This is a spice limitation. The number of parallel devices must be known to the model equations to estimate mismatch parameters. The more the parallel devices the lower the mismatch.

mhajimorad, 2023-11-18:
Thank you @StefanSchippers !  

I will close this thread.

closed 2023-11-18 by mhajimorad