StefanSchippers/xschem_sky130 #19: Simulating sky130_stdcells in xschem & ngspice

mhajimorad, 2023-12-04:
Hi @StefanSchippers ,

Regarding the use of `sky13_fd_sc_hd` standard cell library components in xschem and running ngspice, how do we setup ngspice to find the corresponding subckt definitions.

Upon trying both `TT_MODELS` and `Corner: tt` packages from the xschemrc file, I receive "unknown subckt errors".

Thank you!

StefanSchippers, 2023-12-04:
You don't need to use the `xschem_sky130` repository directly. Install `open_pdks`. This will get the spice models and the xschem symbols.
Detailed instructions are [here](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html).

mhajimorad, 2023-12-04:
Hi @StefanSchippers ,

I had previously installed `open_pdks`.  I do not have any issues simulating circuits with the sky130 devices themselves when I include the path to the model files referenced in `Corner: tt`.

What syntax is needed for the `.include` statement as it pertains to the standard cell library components in `sky130_fd_sc_hd`?

Thank you!

mhajimorad, 2023-12-06:
Thank you @StefanSchippers !

Inclusion of your referenced line in the testbench works.

As a follow-up question:  Are you able to provide some context/background information as to how the constituent transistors of the standard cells were sized?  

Take for example the standard cells `nand2_1` and `nor2_1`.  It appears that all PMOS and NMOS transistors of these gates have been sized `W=1` and `W=0.65`, respectively.  This suggests that transistor sizing was not based-on arriving at comparable (worst-case) rise and fall times.  So what was the sizing based-on?

Thank you!

closed 2024-08-06 by StefanSchippers