StefanSchippers/xschem_sky130 #23: PDK_ROOT environment variable path not found

Ggurukmm, 2024-08-05:
[image]
 I got warning: PDK_ROOT environment variable is set but path not found 
Help me to solve  this

StefanSchippers, 2024-08-05:
`PDK_ROOT` points to a place on the system where the PDK is installed.
On my system `PDK_ROOT=/home/schippes/share/pdk`, and this directory contains the process data:
```
ls /home/schippes/share/pdk
gf180mcuA  gf180mcuB  gf180mcuC  gf180mcuD  scripts  sky130A  sky130B

```
The error message you get is telling you you have set a path on `PDK_ROOT` but the path does not exist on the system. This means the PDK is not installed or is installed in a different place.
Set a correct path for `PDK_ROOT`:
`export PDK_ROOT=/correct/path/to/pdk/data`

Ggurukmm, 2024-08-05:
I set the PDK_ROOT=/correct/path/to/pdk/data in  bashrc but I got again error 
May i change the edit xschemrc file 
I still not install open_pdk this may cause warning

Ggurukmm, 2024-08-05:
I request you to suggest on this error 
[image]

StefanSchippers, 2024-08-06:
It looks linke the open_pdks installation was broken due to connection errors. Delete the git workspace and do the open_pdks install again.

remember that /correct/path/to/pdk/data is not an actual path. You must provide the location of the pdk (on some systems it is /usr/local/share/pdk, but it depends on your system installation).

Ggurukmm, 2024-08-08:
I try to find but can't see any were is I need to to download only pdk file or other any way to find the pdk

closed 2025-01-03 by StefanSchippers