StefanSchippers/xschem_sky130 #24: Symbol error found

Ggurukmm, 2024-08-05:
I sir
If in one XSCHEM in terminal without any directory I got symbol error
[image]

then if got to directory 
.xschem/xschem_library/xschem_sky130 I got no error 
[image]

Ggurukmm, 2024-08-05:
I sir
If in one XSCHEM in terminal without any directory I got symbol errors 
[image]

StefanSchippers, 2024-08-06:
xschem needs to know the directory where symbols are located. Either add this directory to XSCHEM_LIBRARY_PATH or start xschem in the current directory, since current directory is added to XSCHEM_LIBRARY_PATH by the xschemrc file.
If you start xschem from a different directory you should provide a xschemrc file:

xschem --rcfile /path/to/xschemrc.

see this manual page for details on symbol lookup:
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_libraries.html

Ggurukmm, 2024-08-07:
Sorry sir I may not explain problem correctly 
First I try to open XSCHEM in terminal i show in image 
[image]

closed 2025-01-03 by StefanSchippers