StefanSchippers/xschem_sky130 #31: 2 error one is pdk set path not found and simulation error

Ggurukmm, 2025-01-22:
Hi sir 
I had this problems

[image]

After installations od open pdk I got warning and how to find the path please make an video ,sorry for bothering because I am a new one for this

[image]
 Thankyou

StefanSchippers, 2025-01-22:
See this short video

https://github.com/user-attachments/assets/cca50829-d5e4-4f24-adbf-58d85b2b03c9

StefanSchippers, 2025-01-22:
See this video.

https://github.com/user-attachments/assets/ada5a875-06ad-4929-9389-82c1a91767d4

Ggurukmm, 2025-01-22:
[image]

But there is empty

StefanSchippers, 2025-01-22:
So the open_pdks installation failed. This is an open_pdks issue.

- Delete everything.
- `git clone git://opencircuitdesign.com/magic magic.git`
  - `cd magic.git`
  - `./configure`
  - `make`
  - `sudo make install `
- `git clone https://github.com/RTimothyEdwards/open_pdks.git`
  - `cd open_pdks.git`
  - `./configure --enable-sky130-pdk`
  - `make`
  - `sudo make install`

All the Sky130 pdk will be installed under `/usr/local/share/pdk`

Ggurukmm, 2025-01-23:
Error found while doing make in open pdk

[image]