StefanSchippers/xschem_sky130 #35: Symbol error only at xschem_sky130

Ggurukmm, 2025-01-29:
I got an  symbol error only in that path on this path that is .xschem/xschem_library/xschem_sky130

And when I got type fix_symbols in top.scy file is open with no symbol error then I try same for xschem_sky130 not working

StefanSchippers, 2025-01-29:
Sorry, what did you do exactly?

Ggurukmm, 2025-01-29:
[image]
Here I got error like this

Ggurukmm, 2025-01-29:
But top.sch as not symbol error when is write a commend fix_symbols

[image] 

But when I open standard cell there is an symbol error

StefanSchippers, 2025-01-29:
write this command in the xschem terminal and post the result:
puts $XSCHEM_LIBRARY_PATH

do a copy/paste of the text, don't send photos.

Ggurukmm, 2025-01-29:
[image]

After put in xschem terminal I got again symbol error

StefanSchippers, 2025-01-29:
You don't read accurately.
I said:

write this command **AS IS** in the xschem terminal

`puts $XSCHEM_LIBRARY_PATH`

and write here the result.
*DO NOT SEND PHOTOS*

Ggurukmm, 2025-01-29:
puts commend i not working in my terminal can I try echo

StefanSchippers, 2025-01-29:
https://github.com/user-attachments/assets/57ebb944-b293-42ec-b681-1191b761b6a0

Ggurukmm, 2025-01-29:
My pdk file in use/local/share../xschemrc
I done the procedure what you show in video
After that i got error is that : can't read "XSCHEM_SHAREDIR": no such variable

StefanSchippers, 2025-01-29:
open a linux terminal and give this command:

dpkg -l xschem

what is the answer?

Ggurukmm, 2025-01-29:
I got no package found matching xschem

Ggurukmm, 2025-01-29:
One this i noticed the when I shifted to test directory xschemrc file is only in 68 byte and and i original path the xschemrc file is 22k byte size

StefanSchippers, 2025-01-29:
Put here the content of xschemrc file in your test/ directory.

Ggurukmm, 2025-01-29:
#### xschemrc system configuration file

#### values may be overridden by user's ~/.xschem/xschemrc configuration file
#### or by project-local ./xschemrc

###########################################################################
#### XSCHEM INSTALLATION DIRECTORY: XSCHEM_SHAREDIR
###########################################################################
#### Normally there is no reason to set this variable if using standard
#### installation. Location of files is set at compile time but may be overridden
#### with following line:
# set XSCHEM_SHAREDIR $env(HOME)/share/xschem

###########################################################################
#### XSCHEM SYSTEM-WIDE DESIGN LIBRARY PATHS: XSCHEM_LIBRARY_PATH
###########################################################################
#### If unset xschem starts with XSCHEM_LIBRARY_PATH set to the default, typically:
# ${HOME}/.xschem/xschem_library
# <install_root>/share/xschem/xschem_library/devices
# <install_root>/share/doc/xschem/examples
# <install_root>/share/doc/xschem/ngspice
# <install_root>/share/doc/xschem/logic
# <install_root>/share/doc/xschem/xschem_simulator
# <install_root>/share/doc/xschem/binto7seg
# <install_root>/share/doc/xschem/pcb
# <install_root>/share/doc/xschem/rom8k

#### Flush any previous definition
set XSCHEM_LIBRARY_PATH {}
#### include standard xschem devices library
append XSCHEM_LIBRARY_PATH :${XSCHEM_SHAREDIR}/xschem_library/devices
append XSCHEM_LIBRARY_PATH :${XSCHEM_SHAREDIR}/xschem_library
## include directory where this xschemrc file is located
append XSCHEM_LIBRARY_PATH :[file dirname [file normalize [info script]]]
#### add ~/.xschem/xschem_library (USER_CONF_DIR is normally ~/.xschem)
append XSCHEM_LIBRARY_PATH :$USER_CONF_DIR/xschem_library 

###########################################################################
#### SET CUSTOM COLORS FOR XSCHEM LIBRARIES MATCHING CERTAIN PATTERNS
###########################################################################
#### each line contains a dircolor(pattern) followed by a color
#### color can be an ordinary name (grey, brown, blue) or a hex code {#77aaff}
#### hex code must be enclosed in braces
array unset dircolor
set dircolor(sky130_fd_pr$) blue
set dircolor(sky130_tests$) blue
set dircolor(xschem_sky130$) blue
set dircolor(xschem_library$) red
set dircolor(devices$) red

###########################################################################
#### WINDOW TO OPEN ON STARTUP: XSCHEM_START_WINDOW
###########################################################################
#### Start without a design if no filename given on command line:
#### To avoid absolute paths, use a path that is relative to one of the
#### XSCHEM_LIBRARY_PATH directories. Default: empty
set XSCHEM_START_WINDOW {sky130_tests/top.sch}

###########################################################################
#### DIRECTORY WHERE SIMULATIONS, NETLIST AND SIMULATOR OUTPUTS ARE PLACED
###########################################################################
#### If unset $USER_CONF_DIR/simulations is assumed (normally ~/.xschem/simulations) 
# set netlist_dir $env(HOME)/.xschem/simulations
#### if this is set to '1' netlists and simulations will go into a simulation/ folder
#### inside the directory containing the top level schematic. Default: not set (0)
# set local_netlist_dir 1

###########################################################################
#### NETLIST AND HIERARCHICAL PRINT EXCLUDE PATTERNS
###########################################################################
[... 421 lines cut ...]
        }
      }
      {nfet_g5v0d16v0$} {
        if { $w < 5 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.7 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
    } ;# switch
  }
  return $res
}

# open_pdks specific:
# Set variables after ${PDK_ROOT} is known
# This overrides some of the variables set above.

set XSCHEM_START_WINDOW ${PDK_ROOT}/${PDK}/libs.tech/xschem/sky130_tests/top.sch

# allow a user-specific path add-on (https://github.com/iic-jku/iic-osic-tools/issues/7)
if { [info exists ::env(XSCHEM_USER_LIBRARY_PATH) ] } {
    append XSCHEM_LIBRARY_PATH :$env(XSCHEM_USER_LIBRARY_PATH)
}

Ggurukmm, 2025-01-29:
set XSCHEM_SHAREDIR $env(HOME)/share/xschem in this line i chaged to set XSCHEM_SHAREDIR /usr/local/share/xschem

Ggurukmm, 2025-01-29:
[xschemrc.txt](https://github.com/user-attachments/files/18588672/xschemrc.txt)

StefanSchippers, 2025-01-29:
# This is wrong
in the test directory your xschemrc file must contain 

## only one single line

**Please read carefully the instructions in the manual.**  You don't read and do things your own way.
this is what you need to write in xschemrc in your test/ directory:

**source .../.../.../pdk/sky130A/libs.tech/xschem/xschemrc**

of course replace .../.../.../ with the path on your system

Ggurukmm, 2025-01-29:
Is i make changes in xschemrc file

Ggurukmm, 2025-01-30:
no working